hexagon5un/AVR-Programming #8: Undefined reference to initUSART

djsherrill, 2015-01-14:
Hi, I'm using AVR studio and when I compile the SerialLoopback code I get an error "undefined reference to 'initUSART. I copied over the USART.h file to my working directory and I'm not receiving any errors for that file. 
Thanks,
Derek

hexagon5un, 2015-02-14:
Sorry for the late reply.  

Studio does strange things.  Are you using the Studio makefile or mine?  If Studio's, then it won't know where to find the additional headers and code that lives in AVR-Programming-Library/

As you noticed, the easiest way to fix everything with Studio is just to copy the contents of the  AVR-Programming-Library/ folder into each and every project's folder. 

There must also be a way to add additional code/library folders to the project, but I haven't used Studio extensively in like 10 years.

Hope you've gotten it working in the meantime.  Let me know?

hexagon5un, 2015-04-01:
Closing for inactivity.

Hope all worked out. :)

closed 2015-04-01 by hexagon5un

naveengautam002, 2017-05-30:
hi, i am using avr studio 7 to compile the serialloop.c , and got these error and i copied all the USART.h and USART.c file in project folder. 
	
Error		undefined reference to `initUSART'
Error		undefined reference to `printString'	
Error		undefined reference to `receiveByte'	
Error		undefined reference to `transmitByte'	
Error		ld returned 1 exit status

Please help me.