hexagon5un/AVR-Programming #9: Help on USART Communication

gfilios, 2015-03-14:
Hi,

I am working with your book and get stuck in chapter 5 "Serial I/O".

When wiring up the serialLoopBack example (code and breadboard) I receive only garbage on the computer side. 

So i simplified the code, but still gets garbage. Unfortunately your book, does not give any suggestion in the case, something is received, but the received values are not what was expected.

Any ideas what might be wrong?

Here is the Code / Using a simple setup (just one LED, and a FTDI Serial Connection):

ATMEL Code
```
#include <avr/io.h>
#include <util/delay.h>
#include "pinDefines.h"
#include "USART.h"

#define DELAY 100
#define BAUDRATE 9600

int main(void) {
	initUSART();
	uint8_t serialCharacter = 0;
	while (1) {
		serialCharacter++;
		printByte('c');
		transmitByte(serialCharacter);
		_delay_ms(DELAY);
	}

	return (0);
}
```

Python Code
```
SERIAL_PORT = "/dev/cu.usbserial-AH02DXO3"
BAUD_RATE=9600

import serial

sp = serial.Serial(SERIAL_PORT,BAUD_RATE, timeout=5)
sp.flush()

while (1):
	response = sp.read(1)
	print "[" + response + "]" + "[" + str(ord(response)) + "]" 
	print "--------------------------------------------------"
```

Unfortunately I can see, nor the character "c", nor any values incrementing:
This is the Output:
```
[?][128]
--------------------------------------------------
[][0]
--------------------------------------------------
[][0]
--------------------------------------------------
[?][128]
--------------------------------------------------
[][0]
--------------------------------------------------
[][0]
```

what is missing in the implementation, or configuration? Where should I continue searching?

gfilios, 2015-03-14:
Solved: It was the wrong F_CPU value.

Working with the ATMEGA328P it should be F_CPU = 1000000

hexagon5un, 2015-04-01:
Yup!  You've gotta set the F_CPU to whatever speed your chip is running.

Later in the book, you'll find places where we run the chip at 8MHz, and you'll need to make sure F_CPU is set accordingly.  (The Makefiles in my code repo should all work, btw.)

closed 2015-04-01 by hexagon5un