hexagon5un/AVR-Programming #11: Can you clarify the license of this code?

ddaygold, 2015-07-10:
Specifically the Makefile found here:
AVR-Programming/setupProject/Makefile

hexagon5un, 2015-07-18:
Good point.  I'll have to pick a license, I guess.  Yuck.  I'll look into it and get back to you.

Whatever I end up using, it'll be as permissive as possible. 

Any ideas?  I really hate the license blobs at the top of other code -- it's just stuff you have to scroll through to get down to business.

ddaygold, 2015-07-23:
I don't really know too much about what's available, however, I found this site to be pretty straightforward and helpful: http://choosealicense.com/licenses/

What I do know is when there is no grant of license, the code is copywritten by default.

adicarlo, 2016-02-02:
For the most permissive license, simply put the code into the public domain.  I believe all you have to do is state that.

The next step up is something like the BSD or MIT license, but these do involve blerbs...

hexagon5un, 2016-03-23:
I'm horrible.  I finally put MIT on it, because it's easy, it's very open, and it's no hassle.  Thanks all for bringing this point up and putting my feet to the fire.

closed 2016-03-23 by hexagon5un