hexagon5un/AVR-Programming #13: Chapter 12 Footstep Detector - Minor Fix for Serial Comm

TheEJM3, 2016-01-01:
This isn't really a fix, just an alternative for terminals to read the serial data as ASCII instead of bytes:

Original code:
    transmitByte(adcValue - 512 + 127);
    transmitByte((lowValue >> 4) - 512 + 127);
    transmitByte((highValue >> 4) - 512 + 127);
    _delay_ms(CYCLE_DELAY);

Replace with:
    printByte(adcValue - 512 + 127);
    printString("   -   ");
    printByte((lowValue >> 4) - 512 + 127);
    printString("   -   ");
    printByte((highValue >> 4) - 512 + 127);
    printString("\r\n");
    _delay_ms(CYCLE_DELAY);

The serial comm will look like this afterward:
124   -   124   -   125
124   -   124   -   125
124   -   124   -   125
124   -   124   -   125
...

Hope this helps!

-TheEJM3

TheEJM3, 2016-01-01:
Forgot to add...

Elliot, great book.  I've been learning a ton.  I jumped right into learning the Cortex M series and took a step back after finding a lack of structured tutorials.  This is a great introduction to embedded system programming in C.  Fun and hands on.  Thank you.

-TheEJM3

hexagon5un, 2016-03-23:
That's awesome -- both the fix and that you've used the book to springboard into the ARM chips.  Thanks very much for submitting.

closed 2016-03-23 by hexagon5un