hexagon5un/AVR-Programming #16: USART does not work

DarkRabbit19, 2016-02-08:
Hi, can someone help me I'm using AVR studio 7 and when I compile the SerialLoopback code I get the folowing error's:

     Description                                                                    File                   Line
     #error "F_CPU must be a constant value"                                   setbaud.h                    121
     #error "setboud.h requires F_CPU to be defined"                        setbaud.h                    113
     #warning "UBRR value overflow" [-Wcpp]                                   setbaud.h                     235
     #'F_CPU' undeclared (first use in this function)                          USART.c                       24

the code I use is: 
   
   /*
 * serialLoopback.c
 *
 * Created: 31-1-2016 11:28:35
 * Author : Dark
 A simple test of serial-port functionality.
 Takes in a character at a time and sends it right back out,
 displaying the ASCII value on the LEDs.
 */ 

#define F_CPU 100000 UL

// ------- Preamble -------- //
#include <avr/io.h>
#include <util/delay.h>
#include "lib/pinDefines.h"
#include "lib/USART.h"
int main(void) {
	char serialCharacter;
	// -------- Inits --------- //
	LED_DDR = 0xff; /* set up LEDs for output */
	initUSART();
	printString("Hello World!\r\n"); /* to test */
	// ------ Event loop ------ //
	while (1) {
		serialCharacter = receiveByte();
		transmitByte(serialCharacter);
		LED_PORT = serialCharacter;
		/* display ascii/numeric value of character */
		} /* End event loop */
		return (0);
	}

thanks in advance

hessa59, 2016-03-23:
I seems like you have your CPU set at 0.1 MHZ ie #define F_CPU 100000 UL. Is this what you want?

Also I find this error #error "setboud.h requires F_CPU to be defined"  to be strange. Shouldn't it be setbaud.h

hexagon5un, 2016-03-23:
Hi! 

Hessa59 is right -- you need to up your F_CPU define by a factor of ten.  That's an easy typo to fix.

The other possible gotcha with Studio is that you might also need to set a BAUD define as well.  

It has a way to set global compile-time defines like this.  I can't remember off the top of my head what it is.

DarkRabbit19, 2016-03-29:
I tried to do that but it dit not work.
the libery does not work wit Atmel studio in general.
in the and i made my own libery. (dropbox link: https://www.dropbox.com/sh/8ncnrjdd973qy6i/AAAOIanbFtVdW1teYvIutOmOa?dl=0)
the only thing I am wondering is what you use instead of Atmelstudio.

Xavicci, 2016-03-30:
@DarkRabbit19 hola si funciona en atmelstudio7, agrega #define F_CPU 1000000 dentro de USART.c y escribe bien la palabra setbaud.h .
http://es.tinypic.com/view.php?pic=ipzlmx&s=9#.VvvvwVXhCM8 mira la siguiente imagen, espero que te ayude!

hexagon5un, 2016-04-01:
@Xavicci @DarkRabbit19 

Another issue goes over problems with Studio: https://github.com/hexagon5un/AVR-Programming/issues/10

The BAUD, F_CPU problem is solvable like so:
 According to [this website](http://www.engblaze.com/tutorial-using-atmel-studio-6-with-arduino-projects/#setup), Studio calls them "symbols" and they're found buried pretty deep in some menus: Project - Properties - Toolchain - All Configurations - Symbols. 

But the truth is, I didn't intend to get into Studio -- they change it so often that anything I write about using the GUI will be wrong in a year.  The stock compile is with any avr-gcc toolchain.  See the book for more details?  Or LadyAda has a [good tutorial](http://www.ladyada.net/learn/avr/index.html).

DarkRabbit19, 2016-04-10:
In the end I made a other libery that works.
thank you voor your time and the gread book

hexagon5un, 2016-05-11:
Glad you like the book, and I'm sorry about the Studio woes.  I wish I could keep up with them too, but they're a moving target.

closed 2016-05-11 by hexagon5un