hexagon5un/AVR-Programming #18: In organ.c, the variable "wavelength" should called "period" or "timePeriod"

aaronsky12, 2016-04-28:
In the file organ.c in Chapter05_Serial-IO, there is a variable called `wavelength` that is the reciprocal of frequency. However, in every day usage, "wavelength" refers to the the distance between successive crests of a wave (wavelength = v/f, where v is the velocity and f is the frequency), and therefore wavelength is not equal to the reciprocal of frequency. On the other hand, the "period" of a wave is equal to the reciprocal of the frequency: T = 1/f. 

Therefore, to increase code readability, in organ.c, the variable "wavelength" should called "period" or "timePeriod".

rithma, 2016-04-28:
As a sound engineer by trade, I agree that 'wavelength' was a bit confusing at first.  At the same time, however,  a term like 'period' might seem discouraging to the music production kids.   
How about "frequency"?  Its a comfortable word,  and I believe correct by definition?

/Etienne

aaronsky12, 2016-04-29:
Hi rithma, I appreciate your input on this manner. My issue with using "frequency" is that this variable we are talking about is not frequency, it is the inverse of frequency 1/f. That is why I suggested using "period", since the period T = 1/f in engineering.

rithma, 2016-04-29:
I see.   Since the c function determines the duration of each cycle, 'period' would be a good name. 

I need to stop thinking in Layman's terms, now that I'm learning code....

hexagon5un, 2016-05-11:
aararonsky, you're totally right.  (And since I wrote the code a few years ago, I have no idea why I chose wavelength over period -- it's not like I don't know the difference...)

If it's confusing for someone, it's worth changing.  I'll do that right away.

hexagon5un, 2016-05-11:
... and done!  Thanks again.

closed 2016-05-11 by hexagon5un