hexagon5un/AVR-Programming #23: Mistake !

KaiserS0ze, 2017-03-23:
DDRB = 0b0000001;(6 zeroes after b)

should be 

DDRB = 0b00000001; (7 zeroes after b)

its making the wrong pin as output, according to the circuit in MAKE: Avr programming

hexagon5un, 2017-03-25:
Oh crap!  Thank you.  I have no idea how that got through.

Fixed.

Thanks again.

closed 2017-03-25 by hexagon5un

KaiserS0ze, 2017-04-11:
Hey,
It's no problem. I recently purchased your book and I love it. But, I am
having some problems in running the code. If it's okay can I share them
with you? I am currently on the serial communication chapter.

Best wishes

Abhijeet Dutt Srivastava

Closed #23 <https://github.com/hexagon5un/AVR-Programming/issues/23>.

hexagon5un, 2017-04-12:
Sure!  Let me know what's up.

KaiserS0ze, 2017-04-12:
> issuecomment-293194409

Hello,

Thanks in advance !

 In the the program listing serialloopback.c you have included some user
defined libraries like the USART.h. I downloaded this from github and saved
it in the folder containing the code and makefile but still the program did
not run so I defined the specific functions in the program itself. And the
compiler asked me to define the Baud rate and F_CPU and I don't know what
F_CPU means. So as you can see its become pretty cumbersome and I have
totally detracted from learning serial communication. I am also including
the code I ended up with. If you could tell me what to do I would be
grateful.

best wishes

Abhijeet Dutt Srivastava

-- 
Best Wishes
Abhijeet Dutt Srivastava

#include<avr/io.h>
#include<util/delay.h>

void initUSART(void);
void transmitByte(uint8_t data);
void printString(const char myString[]);
uint8_t receiveByte(void);

#define F_CPU 8000000 // I found this value which I simply pasted here 
#define BAUD 9600

#include<util/setbaud.h>

int main(void)
{
char serialCharacter;
DDRB = 0xff;
initUSART();
printString("Hello World\r\n");

while(1)
{
serialCharacter = receiveByte();
transmitByte(serialCharacter);
PORTB = serialCharacter;
} 

return 0;
// Code is not showing any error but I am not satisfied
}

// Functions defined below

void initUSART(void)
{                                /* requires BAUD */
  UBRR0H = UBRRH_VALUE;                        /* defined in setbaud.h */
  UBRR0L = UBRRL_VALUE;
#if USE_2X
void  UCSR0A |= (1 << U2X0);
#else
  UCSR0A &= ~(1 << U2X0);
#endif
                                  /* Enable USART transmitter/receiver */
  UCSR0B = (1 << TXEN0) | (1 << RXEN0);
  UCSR0C = (1 << UCSZ01) | (1 << UCSZ00);   /* 8 data bits, 1 stop bit */
}

void transmitByte(uint8_t data) {
                                     /* Wait for empty transmit buffer */
  loop_until_bit_is_set(UCSR0A, UDRE0);
  UDR0 = data;                                            /* send data */
}

uint8_t receiveByte(void) 
{
  loop_until_bit_is_set(UCSR0A, RXC0);       /* Wait for incoming data */
  return UDR0;                                /* return register value */
}

void printString(const char myString[]) {
  uint8_t i = 0;
  while (myString[i]) {
    transmitByte(myString[i]);
    i++;
  }
}

hexagon5un, 2017-04-18:
F_CPU is defined in the Makefile.  Are you using Atmel Studio?  They have a location somewhere that you can enter user-defined parameters like this.  

> Project -> Toolchain -> AVR/GNU C Compiler -> Symbols
then add F_CPU=8000000 (or whatever)

http://www.avrfreaks.net/forum/how-set-fcpu

KaiserS0ze, 2018-01-24:
Hey Eliot !

I had left working with your book ! I recently started again and am again
stuck on the serialloopback code. Answering your previous mail - No I use
programmer's notepad/Winavr. Please tell me where to get the pinDefines.h
library and the USART.h library. I really love your book and I am learning
a lot, I want to point out something on page 73 where its written "0xFF is
255 in hexadecimal" won't it be decimal? Thanks!

KaiserS0ze, 2018-01-24:
Hey Eliot !

Here is a picture of the error I am receiving. Thanks again !

KaiserS0ze, 2018-01-25:
Hey Elliot I got the code running, soon after I wrote the above mail.
However, I defined the functions you made in USART.h in the program itself.
I use Winavr and Mfile please tell me how to compile a custom library along
with the program- I just can't understand. Thanks so much. P.S Sorry If I
bothered you with too many mails

Abhijeet