hexagon5un/AVR-Programming #25: USART.h header file does not work

naveengautam002, 2017-06-02:
/*
A simple test of serial-port functionality.
Takes in a character at a time and sends it right back out,
 displaying the ASCII value on the LEDs.
*/

// ------- Preamble -------- //
#define F_CPU 12000000UL
#include <avr/io.h>
#include <util/delay.h>
#include "pinDefines.h"
#include "USART.h"

int main(void) {
  char serialCharacter;

  // -------- Inits --------- //
  LED_DDR = 0xff;                            /* set up LEDs for output */
  initUSART();
  printString("Hello World!\r\n");                          /* to test */

  // ------ Event loop ------ //
  while (1) {

    serialCharacter = receiveByte();
    transmitByte(serialCharacter);
    LED_PORT = serialCharacter;
                           /* display ascii/numeric value of character */

  }                                                  /* End event loop */
  return 0;
}
--------------------------------------------------------------------------------------------------------------
***********************************************************************************************************
ERRORS
 
error       "recipe for target 'USART.o' failed	"
error       "F_CPU must be a constant value"	
error      "setbaud.h requires F_CPU to be defined"	
#warning "UBRR value overflow" [-Wcpp]	
error      "F_CPU' undeclared (first use in this function)"
each undeclared identifier is reported only once for each function it appears in	
Severity	Code	Description	Project	File	Line
Error	"recipe for target 'USART.o' failed"

-----------------------------------------------------------------------------------------------------------------
using ATMEL STUDIO 7.0.

baldowl, 2017-06-02:
See note on page 18 about Atmel Studio and the need to define `F_CPU` and `BAUD` at the project level.

naveengautam002, 2017-06-03:
Page 18 of which?

adicarlo, 2017-08-13:
p 18 of the book, _Make: AVR Programming_

skyu74, 2018-10-27:
The following error occurs.
Is there a way around this?
==================================
#error "setbaud.h requires F_CPU to be defined"
#errot "F_CPU must e a constant value"
#warning "UBRR value overflow" [-Wcpp]
==================================

skyu74, 2018-10-27:
> Make: AVR Programming

Can you explain in detail?

skyu74, 2018-10-27:
> Page 18 of which?

Can you explain in detail?

skyu74, 2018-10-27:
> Atmel Studio에 대한 18 페이지의 설명 `F_CPU`과 `BAUD`프로젝트 레벨에서 정의해야 할 필요성에 대해 알아보십시오 .

Can you explain in detail?

hexagon5un, 2018-11-13:
The code sets FCPU = 12 MHz.  (What?)  But then you're using Studio, which maybe ignores this value anyway.  

I have not kept up with Studio in the last few (5?) years, so I'm unable to help there, but the error really looks like you're not defining FCPU on the right page in the project's settings.

goalomy, 2020-11-02:
hey @naveengautam002 first of all remove F_CPU 12000000UL from your main code and #define F_CPU 1000000 UL in USART.c.
here is pic for your reference
[image]