hexagon5un/AVR-Programming #26: Changing baud rate and clock cycle in Make file does nothing (serial not working mega328)

JosiahCochran, 2017-06-08:
I am using the atmega328, which compiles fine if I change the chip to atmega328 from 168p in the makefile, but in chapter05 for the serial, when I hook the output up to a scope, the baud rate is way off and changing it in the makefile does nothing to the output. also, changing the clock frequency does nothing either. I realize the clock is set with the fuses, so Im not entirely sure what the clock frequency portion of the makefile does, but I tried following the header files back to see where baud rate is set and I cant seem to find the difference between the 168 and 328 that would cause the baud rate to be so far off.

JosiahCochran, 2017-06-08:
okay, first off, I was using an rs323 cable, which I fixed usign an op amp and not gate, but my baud rate is 1200 so there is a devide by 8 set somewhere, but I still cant find out how to change the baud rate. serial loopback now works since I did the data conversion from rs232, but my baud rate is 1200. sidenote, I tried the blink code with the attiny85 and it works sometimes, but most of the time it compleatly fails. could you possibly write something on what needs to be changed to use your code for AVR's other than the168p?

baldowl, 2017-06-08:
The fact that your baud rate is 1/8 of expected one makes me think that your ATmega328 is running with the default 1/8 prescaler (is it brand new?)

See pages 189-192 for more details.

JosiahCochran, 2017-06-08:
I read through that section in the book, thatk you! very helpful! I now understand the makefile better and how to actually change fuses and things!

closed 2017-06-08 by JosiahCochran