hexagon5un/AVR-Programming #35: Atmel Studio 7 Serialloopback error USART.H

igonine, 2018-05-02:
Hi:

I known that this issue has already been discussed in previous posts, but after reading all of the posts, i still can´t find an answer to this problem.
The project recognizes the USART.H and pinDefines.h, but when i compile, it gives many error. I checked all the code and can't find what i'm doing wrong.
The errors are in the attached file.

if someone has manged to compile and make it work with atmel studio, please help me so i can progress in the book.

Thanks
João Ferreira
[error.xlsx](https://github.com/hexagon5un/AVR-Programming/files/1968715/error.xlsx)

peterstadie, 2018-05-30:
Same here, I tried many ways to include the files in Atmel Studio 7 but nothing works. I will take a step back and use WinAVR from here on, the IDE just adds at this stage unnecessary complexity and it is much harder to debug.

athulsnair, 2018-11-23:
@igonine  I could compile the code on Atmel Studio 7