hexagon5un/AVR-Programming #37: Chapter 8 Interrupt code and serialScope do not match together

Seikilos, 2018-12-29:
There is something odd in the setup.
The serial scope under https://github.com/hexagon5un/AVR-Programming/blob/master/Chapter08_Hardware-Interrupts/capSense/serialScope.py
does only handle USART data in terms of reading it as `ord` (https://github.com/hexagon5un/AVR-Programming/blob/master/Chapter08_Hardware-Interrupts/capSense/serialScope.py#L6)

But the entire `capSense` code at trunk does not work well with that. It sends string data and newlines to the serial scope making no sense. See [here ](https://github.com/hexagon5un/AVR-Programming/blob/master/Chapter08_Hardware-Interrupts/capSense/capSense.c#L40) and [here](https://github.com/hexagon5un/AVR-Programming/blob/master/Chapter08_Hardware-Interrupts/capSense/capSense.c#L63).

I also spent about two days trying to get this working without success because I am using Atmel Studio with my custom defined header and project files (not the Makefiles), where I've defined `#define F_CPU 1000000UL // 1 MHz`
Since this chapter delays the explanation of the line `clock_prescale_set(clock_div_1); /* full speed */` and does not even briefly mention that `F_CPU` must be adjusted here  I would somewhere drop a hint for newbies like me.

rodolfoacostacastro, 2025-01-01:
I agree here. I think something changed when they updated the program to python3...5 years ago. I have the same issue: capSense.c is sending strings and chars, but serialScope.py is reading just bytes.  I don't know how to solve it, because I'm ignorant in python. I can read the right values using the Serial Monitor in the Arduino IDE as an Arduino Uno board, because there the serial monitor is expecting strings. So I kind of solve the problem, but it would be nice to adjust serialScope.py :)  I´m enjoying the book a lot!

rodolfoacostacastro, 2025-01-02:
Now I'm getting the right values :)) 
Changing this functions on serialScope.py will do the trick for the capacity sensor in chapter 8:

def readValue(serialPort):
    return(serialPort.readline().decode('utf-8').strip())

def plotValue(value):
    print(value)