hexagon5un/AVR-Programming #43: file not found error on first make attempt

tonydavies, 2019-08-03:
Hi, 
I have tried the blinkLED example and get this error on make
C:\AVR-Programming-master\Chapter02_Programming-AVRs\blinkLED>make flash
avr-gcc -Os -g -std=gnu99 -Wall -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums  -ffunction-sections -fdata-sections  -DF_CPU=1000000UL   -DBAUD=9600UL -I. -I../../AVR-Programming-Library -mmcu=atmega168p -c -o blinkLED.o blinkLED.c
process_begin: CreateProcess(NULL, avr-gcc -Os -g -std=gnu99 -Wall -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -ffunction-sections -fdata-sections -DF_CPU=1000000UL -DBAUD=9600UL -I. -I../../AVR-Programming-Library -mmcu=atmega168p -c -o blinkLED.o blinkLED.c, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [blinkLED.o] Error 2

The AVR-Programming-Library is 2 folders above my source/make
The avrdude_utilities is on my search so I can run make from the source folder

From what I remember of make, all .c files in a folder and those in the LIBDIR are linked and compiled to an object file?

I am not sure what this error is telling me. It appears to say it can't find the object, but isn't that what make does, creates the object file, blinkLED.o ?

Any help is welcome!

Tony

Nable80, 2019-08-03:
As you can see, the error message here reads as ``"CreateProcess(NULL, avr-gcc, ...) failed."``

I think your system couldn't find ``avr-gcc`` executable. Most probably you have to manually adjust PATH environment variable (i.e. add path to the directory that contains ``avr-gcc.exe``) ~~or use some better OS where development is not such a pain and paths are managed by package manager in a right way automatically~~.

tonydavies, 2019-08-03:
Thanks, I thought the compiler would be in the avrdude_utilities folder, but didn't check. I found it deep in the bowels of AtmelStudio. The compiler ran, I have a different error now,  which I'll try and solve

tonydavies, 2019-08-03:
Bingo! Chip type was wrong. All good

closed 2019-08-03 by tonydavies