hexagon5un/AVR-Programming #49: Issue with Serial Connection on Arduino

willie-shen, 2021-03-03:
I am currently on Chapter 5. So the thing is I am using an Arduino (meaning using the m328p). I flash my arduino with the SerialLoopback code. I hook up a USB to TTL from my computer onto my Arduino.

It is strange, because when I set the bit rate on the serial monitor to 9600, the LED lights are not lighting up. In addition to that, the serial monitor is printing out unreadable code. 

The only way the LED lights are lighting up is if I set the BAUD rate to 115200. However, I am still getting gibberish data printed out on the serial monitor. Why is it that when I set the BAUD rate to 9600, I don't get any LED lights lighting up (the RX and TX lights on the USB to TTL do happen to flash)

willie-shen, 2021-03-03:
To add onto that, if I connect the RX and TX together on the USB to TTL, I am able to loopback the data

willie-shen, 2021-03-03:
I also noticed that the rated speed, using CoolTerm, was 115200, so I changed the BAUD to that value on the makefile. In addition, I changed USB to TTL serial adapters, but was still getting gibberish output on my laptop, anyway to fix this

bruhbruh5, 2025-05-06:
Did you ever find a fix? I'm experiencing this as well.

bruhbruh5, 2025-05-06:
Nevermind, I found out it's because I was trying to get the AVR to communicate with the RS232 protocol (which transmits and receives -15-15V signals) instead of TTL (0-5V). Probably something the other should've mentioned in the book, but alas.

Side note: You can get the RS232 to work using a level shifter circuit but it's not worth the timer, so just get a usb to ttl adapter instead.