hexagon5un/AVR-Programming #50: littlehacks.org - looks like it's down

merlinblack, 2022-01-21:
Just a heads up!

The home page looks to be a test page  ... "The littlehacks virtual host is barely working!" Other pages are not working.

I'll just have to go back to reading your book - which I am really enjoying. :-)

hexagon5un, 2022-01-21:
Hiya.

Yeah, I really need to get some time to fix that!  (Barely working is better than not at all?)  I think some stuff got misplaced in a server move.

What exactly were you looking for?  I can't even remember what I had up there.  (Sincere apologies, and glad you like the book!)

merlinblack, 2022-01-22:
Oh, nothing in particular - was just curious what was there as you mentioned it in the README of this git repo. :-)